rodizio1/EZ-WifiBroadcast #173: License of the project code

bortek, 2018-10-26:
This is probably the most boring subject but we need to OK it before I can add License document to the project page. I am not the expert on the subject but do have general knowledge about it. And to get even more clarification I asked on this community page some question. you can read it it's not that much text.

https://opensource.stackexchange.com/questions/7538/licensing-current-and-future-change

My conclusion is that it's easiest to stick to GPL2 license and just get over it. GPL2 basically boils down to providing source code together with binaries (which we of course do if we consider an image a binary).

Raspbian on it's own comes under GPL license but it has lots of packages with different licenses but we are still OK to use and modify on top of it as long as we provide the code. BTW anyone knows which Raspbian base image we are using in 1.6 ?

Do we know if we are using any other code which is incompatible with GPL?

Let me know your thought on the subject.

Yes21, 2018-10-26:
> BTW anyone knows which Raspbian base image we are using in 1.6 ?

In the latest building script of @RespawnDespair, the raspbian version he used is : "**2018-10-09-raspbian-stretch-lite**"

RespawnDespair, 2018-10-26:
I'd say were pretty much stuck with GPLv2 (which is fine by me) as stated on stackexchange.

pilotnbr1, 2018-10-27:
+1 on gplv2. 

As for “recognition of contributed/borrowed code”, just a short list giving credit to those who wrote much of the code.. Befinitiv for th original wbc,Anemostec,Rangardid for the OSD, Rodizio, etc
I believe Rodizio already had such a short paragraph in place

I know it’s not an exciting subject, licensing, credit, etc... but these are the things that give it credibility and encourage others to use it and participate. Thanks for taking this on @bortek !

Yes21, 2018-10-27:
+1 for GPL v2.

careyer, 2018-10-28:
Sure thing... GPLv2. Thanks Luke for sharing your thoughts which I can not add anything to.

bortek, 2018-10-28:
thanks everyone for your input. I will add update licens part on the landing page. Overnout.

closed 2018-10-28 by bortek