rodizio1/EZ-WifiBroadcast #175: Feature Request : Add FrSky Telemetry to RC control

Yes21, 2018-10-31:
[FrSky](http://ardupilot.org/copter/docs/common-frsky-telemetry.html) telemetry is now only possible with an FrSky reciever (X6R, X8R, ...) and a FrSky   RC (X9D, X9E, Horus, ...). A special wire with a RS232<->TTL converter is connected between a serial port of a PiwHawck (Passthrough FrSky Telemetry protocol) and the smartport of the X8R reciever. This allows to have all telemetry on the RC screen with a lua script like [https://github.com/yaapu/FrskyTelemetryScript](https://github.com/yaapu/FrskyTelemetryScript).

When one choose to use wifibroadcast RC, he can't anymore have FrSky telemetry on his Taranis.

It would be awesome if somebody could find a way to generate FrSky telemetry from Mavlink on the groudPi, and to send it to the RC usb ports.

careyer, 2018-10-31:
@Yes21 : This is pretty much the same feature request as #125 - only a different LUA script is used ;-). I also believe that this would be a wonderful addition

Yes21, 2018-10-31:
@careyer : sorry, I close it.

closed 2018-10-31 by Yes21