rodizio1/EZ-WifiBroadcast #176: unable to open interface intwifi0 in pcap

Mani17192, 2018-11-01:
Hi,
I am using v1.5 for both TX and RX with Alfa AWUS036NHA  and I am facing a error.
ERROR: unable to open interface intwifi0 in pcap: intwifi0: that device is not up.
But whenever I run the .profile script manually it works like a charm.

careyer, 2018-11-01:
@Mani17192 : Hi.... the latest stable 1.5 is actually very very old (more than a year).... please try the latest 1.6rc6 and see if the problem persists. Thank you.

Mani17192, 2018-11-01:
@careyer , I have used 1.6rc6 and I found v1.5 more stable in terms of video feed quality.
Only this little error is disturbing me. Ill be grateful if can you give me some heads up to figure out the root cause like if I can put "sleep" some where in the .profile because so far I have understand that this is happening due to some service which is overlapping the boot or hardware synchronization.

rodizio1, 2019-01-11:
Sorry, 1.5 is too old, won't spend time on that release anymore. It would be nice if you could open an issue though with your findings about the quality being better with 1.5 compared to 1.6.

closed 2019-01-11 by rodizio1