rodizio1/EZ-WifiBroadcast #177: Incorrect image diagram in Wiki

pilotnbr1, 2018-11-01:
The tx and rx pins on the pixhawk side appeared reversed in the diagram. I don’t own a pixhawk but after some quick googling the pins on the pixhawk side appeared reversed in this image. 

[image]

I removed the image off 2 pages in the wiki. The text was correct in the wiki. Can someone with a pixhawk please confirm the pins?

careyer, 2018-11-01:
No... my diagram is perfectly okay.
Check out the official manual for Pixhawk and the official pinout here:
http://ardupilot.org/copter/docs/common-pixhawk-overview.html

It reads left to right:
VCC | TX (OUT) | RX (IN) | CTS  | RTS | GND 

The image was just misinterpreted by a RCgroups fellow.
May I ask that you undo your change?

Yes21, 2018-11-01:
> VCC | TX (OUT) | RX (IN) | CTS | RTS | GND

On PixHack and PixHawk 3 Pro, that's : VCC | RX (IN) | TX (OUT) | CTS | RTS | GND

See the picture (from me) [here](https://github.com/bortek/EZ-WifiBroadcast/wiki/Software-~-Advanced-~-Telemetry-&-OSD).

careyer, 2018-11-01:
Okay... but we have a Pixhawk in this diagram and this is VCC | TX | RX | CTS |...
It is the most common FC out there right now...
And you can without any doubt believethe official pinout :-)

You can crosscheck here also:
[image]

[image]

pilotnbr1, 2018-11-01:
Ok ill repost the diagram. Pixhawk and clones made this confusing. I will add a large note to the wiki to clarify that the diagram is for pixhawk 1 only. Perhaps we can add diagrams for the other types as well or at least a pinout text.

If I was a new user and bought something other than a real pixhawk 1 I may have made a mistake also and then incorrectly assummed it was a problem with wbc lol

Yes21, 2018-11-01:
> On PixHack and PixHawk 3 Pro, that's : VCC | RX (IN) | TX (OUT) | CTS | RTS | GND

Sorry ! On P3 Pro it's also VCC | TX | RX | CTS | RTS | GND ...

closed 2018-11-01 by pilotnbr1