rodizio1/EZ-WifiBroadcast #180: Uplink Telemetry issues with GCS

Yes21, 2018-11-12:
I create a new issue in order to share our experiences about the mysterious behavior of uplink telemetry.
I did many testings for a week, and had many difficulties to have reproducible results :

- one time only flight mode change was working,
- next time nothing was working anymore,
- some times I could upload waypoints to the Pix, and then delete them,
- but it didn't work anymore the next time !

My different use cases are following : 

- 1.6RC6 : airPi + CSL300 (@5805) / groundPi + CSL300 (@5805)
- 2.0beta : airPi + AWUS036ACH (@5805) / froundPi + AWUS1900 (@5805)
- airPi serial connected to a PixHawk
- groundPi wifi hotspot with an external CSL300 (@5180)
- a Linux PC with QGC (3.4.4) connected either on wifi hotspot, or with Ethernet hotspot
- an Android tablet with QGC (3.4.4) connected either on wifi hotspot, or with USB tethering

In any use cases did the uplink telemetry work at the first time ...
After many tries, I've noticed that I had to reboot the groundPi one or two times until uplink telemetry could perfectly work (changing flight modes, arming, doing calibrations, uploading waypoints) !!

The problem (uplink telemetry never works at the first time) and the "solution" (reboot the groundPi one or two times) are reproducible for me independently of the connection mode (wifi hotspot, USB tethering, Ethernet hotspot), and of the EZ-WBC version (1.6RC6 or 2.0beta).

Is anybody able to reproduce this "solution" ?

zipray, 2018-11-13:
Is this related to "do not set bitrate for Ralink, Mediatek, Realtek, done through tx parameter",except for ath9k_ht drivers?

Yes21, 2018-11-13:
@zipray 

> Is this related to "do not set bitrate for Ralink, Mediatek, Realtek, done through tx parameter",except for ath9k_ht drivers?

Where did you find this advice ?

zipray, 2018-11-13:
https://github.com/bortek/EZ-WifiBroadcast/blob/26514f7d6d7f7ec42e3d16e98a4b164f6bd4870e/wifibroadcast-scripts/profile#L307

Yes21, 2018-11-13:
@zipray  Ok, Thanks.

closed 2018-11-30 by Yes21