rodizio1/EZ-WifiBroadcast #181: Inav documentation and Wiki

pilotnbr1, 2018-11-15:
Need some INav specific examples for setting up easy. I am an INav user and might get to this and have been intending on doing it but time is always my issue...

Additionally I think any graphics should be consistent with what has already been added to the wiki for pixhawk.

careyer, 2018-11-19:
@pilotnbr1 : I am sorry but I do not have any iNAV flight controller so I cannot really help in this regard. However if you need specific graphics feel free to reach out to me. I will then try to put something together in Photoshop then.

pilotnbr1, 2018-11-22:
@careyer I am sitting down to work on the rc INav portion of the wiki (thanks for getting a start on that) and I think the whole rc portion could benefit from some reorganization. I want to get your blessing first as most of it was your work.

My proposal for the rc section pages is-

- General
-ardupilot specific
-INav specific
-Builds

To my there needs to be one general page that makes clear the two routes for rc. Serial or Mavlink. I think here some of the General Notes and warnings @rodizio1 mentioned on rcgroups today can also be described

I think the ardupilot should be condensed and combined into one page.

The INav page I’ll just expand on and work on right now.

Reliable setups or builds should be just builds. While the examples presented are indeed reliable I think they are more demonstrations of what is possible at the extreme end of the spectrum. I look at that page and say “wow” that’s nice! 

Thoughts?

careyer, 2018-11-22:
> 
> 
> I just updated the rc INAV Page. Thoughts? What did I screw up, leave out, etc?
> Please, Jump in and change it if you don’t like it!
> 
> I couldn’t remember about setting baud rate in the INav configurator... it should be mentioned.
> 
> Also @rodizio1 is there a better serial protocol? I happened to choose sumd and it works but others use ibus.. and reports of msp. Should we recommend one of the others?
Thank you very much! You did a decent job. I just changed the "Supported SOCs" link for "Supported WiFi Adapters" since you are refering to that only Atheros AR9271 is supported ;-)

rodizio1, 2018-11-22:
Regarding the protocols: sumd, ibus and xbus as well as msp are all fine from a technical standpoint, doesn't really matter which one. I implemented sumd,ibus and xbus to give flexibility and because they are 'standardized', and probably won't change eventually, like msp did several times (leading to msp currently only working with totally outdated cleanflight versions). msp has one advantage though: Baudrate is not fixed at 115200 and thus you can run it at 19200 with softserial (115200 doesnt work with softserial) in case you don't have enough hardware uarts.

pilotnbr1, 2018-11-23:
Thanks for the help and input guys. Good stuff regarding protocols. One thing I just stumbled across is the potential for ibus to share serial rx and telemetry on one port. So maybe it would have an advantage if wbc could parse ibus telemetry, if I’m understanding that correctly.

 I hope you find the new layout logical. In the RC General page I tried not to be mavlink or INav specific. Just stuff that holds true for both setups.

For the INav page I did add a note regarding telemetry because it seems to be a common use.

Probably can close this issue soon unless more conversation develops..

careyer, 2018-11-23:
Very well done! The RC section looks neat and nice now! =)
Good job on consolidating much of the information on the 'General' page. :+1: 

@rodizio1 : Regarding the regular RC protocols, are all of them supported to their full extent (e.g. max number of channels as of their specification) or are there any limitations or differences between them? If there is it might be worth mentioning this somewhere. I could not find information on this but maybe I haven't looked hard enough. It would be nice if you could tell us here in short. Thank you very much.

closed 2018-11-24 by pilotnbr1