rodizio1/EZ-WifiBroadcast #182: Control and stream from Sony camera via HD Video Audio  Capture device

Egho9, 2018-11-15:
hello everyone . I want use 2 camera on board, first pi camera connected via CSI ,second Sony camera connected via HDMI- USB adapter,last  need control via HDMI. I read all wiki section but nobody dont speak about HDMI to USB HD Video Audio  Capture for take picture from camera and control it. from Sony camera can take stream 1fps ,such do with thermal camera conected  [ https://www.alibaba.com/product-detail/HDMI-to-USB3-0-HD-Video_60305113716.html?spm=a2700.7724838.2017115.1.19de6f8f6YedlH]

Yes21, 2018-11-15:
I'm used to control a Parrot Sequoia from my Pixhawk with this small board :
http://vp-systems.eu/camremote.html
The board is connected to a PWM out of my Pixhawk and to the cam with usb.
I can trigger the cam with my RC, or with QGC missions ...

Hope this help.

rodizio1, 2019-01-11:
Sorry, cannot help on that one.

closed 2019-01-11 by rodizio1