rodizio1/EZ-WifiBroadcast #183: EZ-WifiBroadcast and a realtime kernel

JAR4x4, 2018-11-18:
Hi guys! 

I'm currently building my own flightcontroller based on this work:
https://discuss.ardupilot.org/t/building-an-autopilot-from-scratch-using-a-raspberrypi-zero-mini-zee/11722

Currently my setup is based on two PI 0's one running the ardupilot code with the sensors connected and one running EZWB for telemetry, rc control and videolink. I have not tested the setup yet, hopefully I will be able to benchtest the setup during the next week.

I would really like to ditch the second PI 0 and run everything of one PI 0. Is it possible to patch the EZWB image with a RT-PREEMPT kernel like this:
http://www.frank-durr.de/?p=203

?

Alex

rodizio1, 2018-11-19:
quoting myself from here https://www.rcgroups.com/forums/showpost.php?p=39512138&postcount=4065

> Problem wih that Emlid image is, it needs a realtime kernel. I've experimented with realtime kernels on the Pi and Wifibroadcast some time ago, the short answer is: Forget it, these kernels just don't work right on the Pi with wifibroadcast and usb wifi cards and everything. Tried all kinds of different kernel and cmdline.txt settings, sat there testing and compiling for days. Either there is packetloss on the usb bus, high cpu load, instabilities, or a combination of everything. Yes, I know emlid supplies a ready-made image with wifibroadcast in it, but I bet 100 bucks that this doesn't work right.
> 
> What I still don't get: Why do you guys want to use those emlid boards or the Pi as flight-control? I mean, running Linux (which is a non-realtime OS to start with) on top of the Raspberry Platform (which makes Linux even more "non-realtime" because of the underlying firmware that does things that are out of Linux's control ...) just doesn't seem to be a good idea for stuff like flight-control which requires guaranteed latency (Which Linux cannot really provide, especially not on the Pi). This has "problems" written all over it.

JAR4x4, 2018-11-22:
@rodizio1 
Noted, and thanks for your reply!

closed 2018-11-22 by JAR4x4