rodizio1/EZ-WifiBroadcast #184: Force TX mode v1.6RC6

JAR4x4, 2018-11-22:
Hi again.

I'm testing my setup now, but have not received my Pi0 camera cable yet. Is it possible to force the TX pi into TX mode without a camera connected. Why? Because I want to use telemetry over WBC

Alex

d3nyys, 2018-11-22:
Theoretically you could change the code in /root/.profile . Search for CAM= and fix the numbers. You also need to make the image writable with 'mount -o remount,rw /'  . I didn't try so I don't know if you can do that.

JAR4x4, 2018-11-22:
Thank's @d3nyys, that worked!

Is it an idea to let this be user configurable?

Alex

Howie9600, 2019-07-11:
I have tried this and it works perfectly.  You need to change two CAM= (data) to CAM=1.

This makes it fixed TX

VU2ATX, 2019-07-18:
> CAM= (data) to CAM=1.

Hi, in which file, I need to do this change?

Howie9600, 2019-07-18:
/root/.profile

VU2ATX, 2019-07-20:
Hi, what else I need to do to enable only usb camera rather than regular pi cam and get the video stream in OpenHD at ground pi ?
I had posted this question in RCG forum of OpenHD for Justin, but did not hear from him looks like, he is tester only.