rodizio1/EZ-WifiBroadcast #185: Make RC work with RTL8812/14au chipsets

Yes21, 2018-11-23:
@rodizio1
I have many questions about wifi adapters RC compatibility :

    should only the wifi groundPi adapter be "compatible" ?
    are the only prerequisites CTS protection and RTS frames injection ?
    ( what means CTS and RTS ?)
    how to know if a wifi card is CTS/RTS compatible ?
    could you explain exactly what is needed to be done for making work RC with a particular wifi board ?

I'm asking all these questions, thinking of RTL8812/14 boards ...
I'm pretty sure they are RC able, and I hope you will point us into the right way to make them work for RC.

Many thanks to you.

Yes21, 2018-11-26:
An user did report [here](https://www.rcgroups.com/forums/showpost.php?p=40519743&postcount=75) that he could get RC control working with AWUS036ACH.

That's very strange, because it's not supposed to work with this card ...

I've never tested it with my 036ACH and will do it tomorrow, but I can confirm I never could have RC working with AWUS1900.

zipray, 2018-11-30:
It's just that mavlink doesn't work.

Yes21, 2018-11-30:
Mavlink telemetry works pretty well with8812/14au.

closed 2018-11-30 by Yes21