rodizio1/EZ-WifiBroadcast #187: Destroying Wiki

pilotnbr1, 2018-11-23:
So what’s the story? 

Is it the wiki going to be restored? Any safeguards so this doesn’t happen again?

Really poor behavior on several fronts today.

bortek, 2018-11-24:
Oh just a little drama with one member not coping with the other and member and overreacting. Frustrating, but people are different. 

Wiki was not the whole story. The landing page was changed too but that was easy to restore.  Im working on restoring wiki which is harder.

I will create a daily remote backup for the wiki later when everything is back.

bortek, 2018-11-24:
Wiki is now restored too.

pilotnbr1, 2018-11-24:
Looks like all image links are broken.

bortek, 2018-11-24:
Not all images but yes some are removed and some are in the other directory perhaps it was renamed. Strange though I have reverted all the deletion commits but the images are still not there. Or alternatively github cannot revert the images as they are on the githubusercontent.com part of the platform. But they are in the caches in different paces so it is recoverable in some way.

Yeah he did it, I hope he did not mean it and that it could have had such consequences.

careyer, 2018-11-24:
Let me know which images are missing, I will send them to you. I would rather like you to respect my wish not to use images I solely created myself however. Sorry for the inconvenience caused. I just tried to be compliant in a way so that our master can finally find piece of mind.

bortek, 2018-11-24:
Thanks for joining in here. I have restored a bunch of it now and there is one more dir left. I will let you know if there will be I could not find.

Yes21, 2018-11-24:
Dear Boris and Luke,

I prefer to use this closed issue to explain you my thoughts, better than to argue on the RC forum.

Yesterday was really the Black Friday !!

First of all, I want to thanks you Boris for all the time you loosed in rebuilding the wiki. I know that was a bad idea to try to destroy this common work. Yesterday when I sent a kind support mp to @careyer, I had no idea what he did to the wiki.
I also apologize because, in a moment of rage, I did also delete 2 or 3 files on the root's github. There were logo files that didn't have nothing to do here. But I never had the intention to destroy my own or others contributions.
So I'm very happy that the wiki is here again, thanks to you Boris !

And now let me say what I think about yesterday's crisis. 
Since the return of @rodizio1 I knew this should arrive, because of the very anti-social behavior of him : he's never happy, always criticizes everybody, is always the one to be right, never says "thanks", is always looking to the past, ... I can't anymore bear this attitude of a guy who's unable to work with others.
Other than adding his negative spirit and attitude to the project he has not really contributed anything meaningful since he returned.
Now I guess he will return to his loneliness, and will preserve us from his blackness ...

We now that we can have a good 2.0 version without his help ...

Thanks again for the wiki.

Yves

bortek, 2018-11-24:
Thanks for chiming in Yves. I am easy on this, and I appreciate your concerns. I wonder why in the name of Saturnus rings everybody are so anxious about rodi?!? He is neither a god nor a boss, just a contributor who did a lot in the past. There is no boss here and everybody are equal and should contribute on equal rights.

It is fine to disagree and argument and discuss but when you still cannot agree on a point go ahead and for example make a vote on a solution/decision. two or more against one and you know who wins. We have done this in the beginning and it worked fine. Honestly, you could have just ignored him and made your point heard, agreed between you two and/or the others and just continued.

As for the Wiki. I am almost through with it now, found some broken links that were there even before as it looks. Will post some results soon when I am done.

careyer, 2018-11-24:
Boris, I hope the backup helped to restore everything. If it did not please reach out to me and I will try to get it fixed. You have done too much unnessesary work already and I am sorry for this. Usually it takes a hell lot to upset me but yesterday it just got over the top. I am sincerely sorry for the way I reacted - I was too deeply offended and more than in only one rage. 
I agree with you that everyone should be equal and that there are unwritten laws for treating each other, especially when there are different views on certain points. At any time and in any place there are rules and limits in dealing with each other that have to be adheared to. Yesterday my line of toleranceline was certainly crossed.
In order to cool down a bit I decided to enjoy the positive attitude of the people here in the Caribbean and relax for the next two weeks. The positive attitude and state of mind of the people here feels so good. I'm not going to extend my Wifi access on the cruise ship for the next 2 weeks (it's too expensive anyway). Should there be any problems with the wiki after that I am willing to get everything rebuild and then say goodbye as I can easily comprehend that I am not a (trust-)worthy member of this team any more. I will be happy to run away and let you continue your work in peace.

Cheers my friends! #T.Sunrise
Thomas

bortek, 2018-11-24:
Thomas make sure you disconnect and take time off. Zoom out and chill out ;) when you're back then get in touch if you feel for it.

pilotnbr1, 2018-11-24:
I think now in a lot of ways as Bortek said it is not just one person who owns it and runs everything. This is open source, collaborative, and community property as such democracy prevails.

Don’t let this thing consume you. We spend our time and energy here and get invested in it and so it’s understandable for some emotion to get involved. Just keep in perspective. 

As far as trusting people I really don’t think that is an issue. The code is most important and it will always be preserved (especially with image builder) as well as frequent backups of everything. Take some time off careyer, get refreshed and I would welcome your energy and efforts again.

bortek, 2018-11-24:
Think I am done with the recovery and to be on the positive note I've found some broken links that had to be corrected. Consider it as a minor overhaul without affecting the content or structure. Not that I spend a hole day on it, dont worry, I was out with my one kid almost all day :)

For those of you who are on the shore, need your input on this particular item. 

Second part of this page 
https://github.com/bortek/EZ-WifiBroadcast/wiki/RC-~-RC-with-Ardupilot#flight-controller-serial-port-setup-via-mission-planner

is I fully redundant to this whole page

https://github.com/bortek/EZ-WifiBroadcast/wiki/RC-~-FC-Parameters

Which one should we keep? I think it is sufficient to have it on one page (the first page?) but want to hear your opinion. 

There are other pages that link to both of these pages but I will fix them when we decide which one to keep.

pilotnbr1, 2018-11-24:
I say keep the first (bigger) page. I think it’s all been condensed into one. With this wiki I am a big proponent of keeping it condensed and EZ ;)

Yes21, 2018-11-24:
I agree.

RespawnDespair, 2018-11-25:
Well, since the dust seems to have settled and hopefully people have had a chance to reflect on the past days now is a good time for me to chime in. I have intentionally not involved myself with the proceedings of the last few days. I have read everyone's comments and have taken some time for myself to determine how i feel about this. Below i have summed up what i've seen over the last year and how i feel about that.

- @rodizio1  has no filter. He is very agressive and abusive in his communication. He is very smart and knowledgeable, but that does not outweigh the negative consequences of his behaviour. I would love to know WHY @rodizio1 behaves in this manner, please tell me. It is quite possible i missed the events leading up to this, but as it stands now you look like an ass, and i'm pretty sure you're not.
- @careyer has been very helpfull in rebooting this project, he takes a lot of time to put information out there, which is, mostly very helpfull. I have however also seen the communications between @careyer and @htcohio where i find @careyer to display a behaviour similar to @rodizio1 . Ultimately causing the split to Open-HD by @user1321 and @htcohio.

I will not takes 'sides' in this as i do not want there to be any sides at all. I feel like i'm stuck in the middle of a high-school drama when all i want to do is code and make this project work.

We project an air of utter incompetence in the rcgroups forums by openly playing out these squabbles, that is unacceptable and totally unprofessional. 

'Art is emotion' is what a friend keeps telling me, and i think there is little difference between our 'art' and the paintings he makes. So i understand emotions running high, which shows passion and commitment to the project. But this needs to be done in a civilized manner.

Bottom live for me:
I am not the biggest contributor to this project, although i like to feel that, in my own way i add to this. 
Never will i remove anything i have already created, since that goes against everything open-source means to me.
**But EITHER the parties involved refrain from this childish behavior OR i will stop any and all involvement with the EZ-WFB project**. I'm on the fence as it is right now, but i don't want to decide anything rash.
As stated i'm here to create something beautiful in my limited free time, i do not want to be confronted with these levels of drama every time i open my inbox. Behave as adults, professionals even, or take it away from rcgroups and Github.

rodizio1, 2018-11-25:
To give some background regarding the 722N's PCB antenna and the recommendation to disconnect it:

Back, when there was only Befinitivs image, as well as with the first EZ-Wifibroadcast versions, there were two big issues which caused pdeepacketloss and badblocks, even in good conditions, where theoretically absolutely no packetloss should occur:

1. That strange Kernel/USB/whatever-it-is packet jitter problem which caused packetloss with multiple RX cards, especially with Atheros (see video on my Youtube channel about it)
2. The packetloss caused by the PCB antenna on the 722N sticks (unfortunately, about this issue, I made no video)

Thoroughly analyzing, understanding, reproducing and in the end fixing/mitigating/working-around those issues took many many many hours of work. Reading a lot of stuff on the internet, doing a lot of tests inside on the desk, compiling kernels, trying firmwares, kernel options, looking at wireshark traces, buying more wifi sticks, USB hubs and Raspberrys for testing. Asking more knowledgeable people like the ath9k-htc-firmware guys about it. Doing tests outside under real conditions with walking around on fields, weekend after weenkend, sometimes walking around for hours trying and confirming different things.

These two things are actually what made EZ-Wifibroadcast working correctly (that is, zero packetloss from being right next to the receiver down to around -85dbm being far away from the receiver, see the videos on my Youtube channel). _All_ other images floating around on the forums at that time didn't have these things fixed and didn't work correctly (And BTW. some of the new images floating around now probably have re-introduced issue number 1 due to untested and not correctly configured kernels).

So basically, that PCB antenna thing is one of the two main pillars of EZ-Wifibroadcast's success: An absolutely clean image, people being able to fly 25km and back without a single visible glitch, as well as an absolutely reliable R/C functionality with very low packetloss on the R/C link.

After I have analyzed and fixed issue number 2 by disconnecting the PCB antenna, I wrote it on the forum. Then waited until somebody reported back about it. Trebor Link did confirm it (and I know his testing is thorough). Then, _after_ I had not only analyzed and reproduced the issue myself a lot of times, but a trustable user in a different country far away also confirmed that reception is better and packetloss is lower with the PCB antennas disconnected, only _then_ I have added the recommendation of disconnecting the PCB antenna to the Wiki.

Now - After repeatedly:
- Complaining about the bad documentation and infos being buried in the long thread, leading him to not being able to make this or that modification (while other users like user1321 or lelik and many others just make their modifications ...)
- Complaining about that shared memory thing he didn't understand and also refused to understand (and then apologizing for it)
- Posting misinformation about r/c and serialport use (and then apologizing for it)

careyer writes this (https://www.rcgroups.com/forums/showpost.php?p=40311113&postcount=6039) on the forum:

>P.S: We also tested the TP-Link 722N v1.10 again. Back then there was the recommendation that the second signal train to the PCB antenna should be deactivated. We have thoroughly tested this myth and I'm afraid but there's nothing to it.Quite the contrary: It is even beneficial to solder on a propper 2nd antenna here which will result in better reception and less packet-loss. No issues with the the switcher and combiner in the chipset whatsoever, hence: NOT "broken by design" 

And declares my finding (remember, one of the two pillars of the absolutely clean picture of EZ-WBC) as "myth", that there is nothing to it, and that it's actually the other way around:

- Without disclosing how he tested or anything
- With a strong confidence/authority in his writing that is staggering
- Without having searched through the thread thoroughly and without knowledge of the ath9k-htc-firmware Github issue
- Without having the required knowldedge about Atheros cards, Linux, or Wifi
- Without maybe first writing something on the forum like: "Hey Rodizio/other users, I tested with the PCB antenna and I couldn't reproduce your findings, here's how I tested, how did you test, can anybody shed some light in this or has done similar tests etc?"
- Without having his findings confirmed by other users first

- And he even deleted the recommendation to disconnect the PCB antenna in the Wiki and replaced it with his recommendation

I write him a PM (to keep this kind of stuff out of the public) about it, he tells me how he did the tests, it turns out they were _not_ done properly and thoroughly and are in no way universal. I also vent about his other behaviour, he agrees about that and that his tests were not universal, apologizes and says we should be friends again. I say okay piece and think all is good again.

Then, just a few days later, he writes this (https://www.rcgroups.com/forums/showpost.php?p=40509961&postcount=6782):

>This is a difficult question that we are still investigating on. Whereas the (secondary) PCB antenna on the 722N can cause the described badblock problem under certain circumstances (starting at an RSSI of about -70dbi and worse). On the other hand it was observed that it can noticably enhance the stability of the link when a proper antenna is connected to th [... cut ...] er sentence contained wrong information. Same thing with the PCB antenna thing, the important info to disconnect the antenna got deleted and replaced with "advice" that could lead to packetloss and video, telemetry and R/C reception issues with 722N sticks.

Just to make this clear: I like most of the stuff he did to the Wiki (and also told him that via PM). Me writing that his improvements are not improvements didn't mean _all_ of it, just the things that are wrong and/or potentially dangerous.

RespawnDespair, 2018-11-25:
@rodizio1 Now that's a thorough response. It sheds some light on the matter.
I agree it is better to give no information than bad information. I do appreciate the things @careyer did for this project and could do in the future.
As to yourself, there is nobody who has more knowledge as to what makes the system work than you, the issues we have with the current image can only be fixed by your input i think.
I don't want to forbid anybody from contributing to the project, how should we move forward?
For the code we can make a simple gated-checkin, any pull request would have to be audited. If this could be done for the wiki as well, we could prevent wrong or incomplete information from showing up as a team.
As for our communications on the rcgroups forums i think we should be more careful. I understand your intent and frustration, but expressing it there will only damage the image of the project.

I for one will absolutely accept you position in regards to technical matters and accept your judgement on these matters. In regards to project management and the image-builder method i rely on myself and appreciate the input of others.

I we all agree to some code of conduct where we are careful not to post unintentionally wrong information by verifying it with the rest of the team first would that work?

I want to get back on track to get the 2.0 image up to the same standard as the 1.6RC6 so we can (together) rewrite the initialization and improve the system further.

pilotnbr1, 2018-11-25:
Thanks for responding @rodizio1. I know quality and accuracy are important to you and is evident in your communication and what you produce. I think we all appreciate that, especially when it comes to RC as it is playing with fire. 

I think a lot of this highlights why a comprehensive and accurate wiki is important. Fewer “what’s this” and “how does this work” questions on rcgroups eliminates the chance for wrong,misinterpreted, miscommunications. WBC has in the past attracted lots of tinkerers and experimenters, but I think and hope it will change more towards hobbyists that choose EZ-WBC for simplicity, reliability, and capability for low cost. So again I really believe in solid documentation (and more content as required) on the wiki. 

@RespawnDespair thanks for your commitment to open source and good conduct. I think it goes without saying that no one wants to lead someone else the wrong way with misinformation. I also agree that having all this very public drama looks terrible (this is not very private either). I wish it was just limited here or these things settled elsewhere. Maybe some sort of google hangouts or something, I am sure everyone would behave better if they were face to face when disagreements come up. 

I too just want to be part of something that is fun, collaborative, and just plain cool for this hobby! I don’t want it to be work (ughh hashed thru that) or drama. So maybe we can get back to what we enjoy and let the rc group drama be buried by the next 100 pages lol

Yes21, 2018-11-25:
> ... and let the rc group drama be buried by the next 100 pages

;-) That's exactly what will append ! 
The drama will fall into this enormous memory, where nobody can find anything except all and its contrary ...
This github, with its wiki and issues, is the best thing we have to develop an manage such a nice project. 

I'm also sure that my [latest posts from a dreamer](https://www.rcgroups.com/forums/showpost.php?p=40520177&postcount=156) on OpenHD RCgroups will disappear without any reactions.
I wanted to share these thoughts with you all here ...

And now we could perhaps reclosed this issue, and go back to next steps ?

bortek, 2018-11-26:
We are all good at things each of us does and that's the strength we have as a project. At the same time we all have different backgrounds, culture, temper and personalities. Let us try use that as a strength in a creative way. Think about others when we communicate and try keep a perspective.

Reason for develop branch being so open and not requiring PR was to simplify work and speed up the process instead of waiting for someone to approve. But of course if you think it is necessary I can enable it requiring all changes to go via PR? That will either require creating extra branch or forking to own user and making thereafter a PR to to this project. This is unfortunately not possible for wiki, even though it is internally versioned. 

I will implement external backup solution today/tomorrow to two independent storage endpoints and afterwards will close this issue.

bortek, 2018-11-26:
Would a slack chat (closed one) be of interest for internal discussions, sharing ideas and (dramas 😊) instead of rcgroups? I have been using it for other stuff and thinks it's awesome.

RespawnDespair, 2018-11-26:
We have a Gitter already running:
https://gitter.im/EZ-Wifibroadcast/Lobby

But it isn't exactly closed...

rodizio1, 2018-12-03:
Regarding drama: It's not me who created the drama. I did not write several subsequent posts on RCGroups, did not vandalize the Wiki, did not create this topic on Github, did not add my two cents after it has been closed already, also, I did not re-open the issue. I simply told Careyer what I think of his stuff. Yes, it wasn't nice, but this problem has a history, as I've explained above, he really asked for it. It was you guys who created the drama and all the fuss.

And it was you guys who actually forced me to explain the thing in public because you started attacking me personally in the public. What exactly was it, that was preventing you from talking to me via PM on the RCGroups forum first?

>But EITHER the parties involved refrain from this childish behavior OR i will stop any and all involvement with the EZ-WFB project.

You tell me about acting like an adult, while at the same time pulling off one of those pathetic "or else" threats? Seriously?

>Behave as adults, professionals even, or take it away from rcgroups and Github.

Again, you, as a newcomer to the project who didn't really contribute anything, are trying to tell me what to do? While it is actually you who is spreading the issue from the forum to Github? 

Yeah, well, ahem, uhm, nope. 

>I don't want to forbid anybody from contributing to the project, how should we move forward?

I decided to move forward like this:
- The project was transferred to my Github account
- The removal of @Yes21 and @careyer already done by Bortek (after they vandalized the Wiki) will be permanent. I have no respect for people who do stuff like that and also cannot trust them anymore
- Your collaborator status will also be removed, @RespawnDespair. As soon as I see you actually contribute to the project (see below, if you still want to), you can have your collaborator status back
- The CoC will be changed to a more merit-based approach. Respect, trust and seniority must be earned through quality contributions, just like in all the other opensource projects.
- Just like in other projects, contributions are accepted as Pull Requests and will be reviewed by the project lead, i.e. me. After a few good pull requests, I will consider adding people to the collaborator list

RespawnDespair, 2018-12-03:
Since this is in the open already, i don't see how replying here would make matters worse.

Regarding the perceived 'threat' of an 'or else statement' i think i didn't explain myself correctly. Whatever you might think of my contributions to this project, i _was_ having fun doing what i was doing. It stopped being fun when people started acting like children. Whether or not YOU were right or not does not change the fact that long before i posted anything visible to others YOU were posting messages that were worded aggressively. Apparently there was a whole history preceding those messages, but they were not public. So what everyone including me were exposed to was only what was posted publicly. My 'threat' is useless, you will probably continue as you have always done without my contributions, it was a heartfelt cry for people to stop ruining something that could be so much fun to work on.

As for behaving as adults, i really tried to keep a mellow tone, knowing a lot of emotion is involved. I really hope you will revisit the things i posted some time from now and see them as they were intended.

For now i will follow my heart and focus my energy on things that make me happy again.

Yes21, 2018-12-03:
@rodizio1 And now, what are your projects for the future of EZ-WBC 1.6RC6 ??

closed 2018-12-04 by rodizio1