rodizio1/EZ-WifiBroadcast #188: relay mode no work

roque-canales, 2018-11-26:
Hello Guys,
EZW 1.6rc6

setup:
air pi zero awus036nha 2484
relay pi 3b with two awus036nha 2484 rx and relay on 2412
groud pi3b with one awus036nha 2412

Connecting screen on relay pi, we can see video and telemetrie (osd)

Connecting screen on ground pi we only se telemetry osd, no video black screen.

How to make it work?

Thank you

regards.

rodizio1, 2018-11-29:
I haven't got time to test yet, but since every single report from you turned out to be a real bug, I guess it's a bug.

roque-canales, 2018-12-10:
I will try with 1.5 for see if this issue is du to 1.6

roque-canales, 2018-12-18:
I check with EZ WIFI Broadcast 1.5 and it works.
I'm not devloper, but could you please check for port this option to 1.6?

rodizio1, 2018-12-18:
Thanks for testing and reporting back. I think I found the issue.

In the .profile script in the rx_function the tx_rawsock for forwarding the stream is started without first initializing the sharedmem for the tx_rawsock:

```
        if [ "$RELAY" == "Y" ]; then
                       ionice -c 1 -n 4 nice -n -10 cat /root/videofifo4 | /root/wifibroadcast/tx_rawsock -p 0 -b $RELAY_VIDEO_BLOCKS -r $RELAY_VIDEO_FECS -f $RELAY_VIDEO_BLOCKLENGTH -t $VIDEO_FRAMETYPE -d 24 -y 0 relay0 > /dev/null 2>&1 &
        fi

```

to fix it, add the sharedmem_init_tx like this:
```
        if [ "$RELAY" == "Y" ]; then
            /root/wifibroadcast/sharedmem_init_tx
            ionice -c 1 -n 4 nice -n -10 cat /root/videofifo4 | /root/wifibroadcast/tx_rawsock -p 0 -b $RELAY_VIDEO_BLOCKS -r $RELAY_VIDEO_FECS -f $RELAY_VIDEO_BLOCKLENGTH -t $VIDEO_FRAMETYPE -d 24 -y 0 relay0 > /dev/null 2>&1 &
        fi
```
Or copy over the attached .profile script file to /root/.profile (mind the dot in front of "profile")
[profile-relay-fix.txt](https://github.com/rodizio1/EZ-WifiBroadcast/files/2691999/profile-relay-fix.txt)

Please let me know if it works (or if you need help making the changes).

roque-canales, 2018-12-20:
Hello Rodizio,

I test this Morning and I feedback.

Rgds

roque-canales, 2018-12-20:
It works!!!!!!!!

roque-canales, 2018-12-20:
Thank you rodizio,

Could you please help me to find how to forward rssi for the first link to the second instead of view rssi off the second bridge?
and how to desactivate completely the OSD for the relay?

rodizio1, 2018-12-20:
Disabling the OSD could either be done by modifying the .profile script (look into osdrx_function) so that it is not started at all, or you could disable all elements in the osdconfig.txt (then it will still be running, but not display anything).

Forwarding the RSSI data is not trivial, this would require writing C code, I can look into it sometime, but it will take a while.

roque-canales, 2018-12-22:
Hello rodizio

Could you please help me to found how to mod .profile to output bidirectionnal telemetry on rx tx pin of the ground pi (assuming that udp telemetry will be disabled for do this)?

I wish you a merry christmas

zipray, 2019-02-12:
> It works!!!!!!!!
I also modified the configuration file with this code. The MAC address is exactly correct, but I kept getting the hint that“Relay card $RELAY_NIC not found!”.

rodizio1, 2019-02-13:
do you have the mac address in all lower-case and without spaces, colons or dashes etc?

zipray, 2019-02-13:
> do you have the mac address in all lower-case and without spaces, colons or dashes etc?
@rodizio1 
Yes,i'm ll lower-case and without spaces, colons or dashes.

like this:
RELAY=Y
RELAY_NIC=00156d6091d1‎
RELAY_FREQ=2484

rodizio1, 2019-02-13:
It seems to be "not there" when the relay card configuration takes place. But a few seconds later, it gets detected (and configured as normal RX card since setting it up as relay card didn't work).

Can you please set DEBUG=Y in the wifibroadcast-1.txt and post the debug.txt file here?

zipray, 2019-02-13:
> It seems to be "not there" when the relay card configuration takes place. But a few seconds later, it gets detected (and configured as normal RX card since setting it up as relay card didn't work).
> I'd say either it's not detected because it's really "not there" yet when the relay configuration takes place (because loading the driver etc. took longer) or it's "not there" because it was there, but got disconnected and reconnected due to power issues.
> Can you please set DEBUG=Y in the wifibroadcast-1.txt and post the debug.txt file here?

[debug.txt](https://github.com/rodizio1/EZ-WifiBroadcast/files/2860532/debug.txt)

rodizio1, 2019-02-13:
I see one USB device disconnected:

`[    3.131746] usb 1-1.5: USB disconnect, device number 6`

But this was a memory stick I think (?)

So the only explanation I have currently is, that the card was not "yet there" when the script tries to look for it. Can you please try insert a few seconds of delay in the detect_nics function in /root/.profile?

Change the script from this:
```
function detect_nics {
	tmessage "Setting up wifi cards ... "
	echo

```

To this:
```
function detect_nics {
        sleep 10
	tmessage "Setting up wifi cards ... "
	echo

```
 This should give the card an additional 10 seconds to be detected/set-up before the script tries to use it as a relay card.

zipray, 2019-02-13:
> sleep 10

The code was patched and tested, which unfortunately only delayed the appearance of  "Setting up wifi cards ... "

rodizio1, 2019-02-14:
hmm, I'm running out of ideas, this is really weird, I'll try to reproduce it myself.

zipray, 2019-02-14:
> hmm, I'm running out of ideas, this is really weird, I'll try to reproduce it myself.
@rodizio1 
I modified the script file to use MAC_RX[3] as a relay wifi card.Then a miracle happened.Please refer to the attachment.
[debug.txt](https://github.com/rodizio1/EZ-WifiBroadcast/files/2865452/debug.txt)
[.profile.txt](https://github.com/rodizio1/EZ-WifiBroadcast/files/2865458/default.profile.txt)