rodizio1/EZ-WifiBroadcast #189: 关于使用第三方相机的疑问

2357531789, 2018-11-29:
树莓派都有USB接口，为什么不考虑使用USB采集卡的方式来支持HDMI信号或者av模拟信号哪？wiki里推荐的Auvidea B101存在兼容性问题，而且成本较高¥800左右，而HDMI转换USB则需要¥300左右既可以获得兼容性很好的设备，甚至只需要30¥就可以获得av模拟信号转换usb的设置，
我想知道是什么障碍导致没有这样选择。。。。

Yes21, 2018-11-29:
Translation :

Raspberry Pi has a USB interface. Why not consider using a USB capture card to support HDMI signals or av analog signals? The recommended Avidea B101 in the wiki has compatibility issues, and the cost is higher than ¥800, while the HDMI conversion USB requires ¥300 or so to obtain a well-compatible device, even if only 30¥ is needed to obtain av analog signal conversion. Usb settings,
I want to know what obstacles lead to no such choice. . . .

bortek, 2018-11-29:
I think this has been up for discussion couple of times. We use pi camera due to low latency and are not considering any other options as they will not be fast enough. All that conversation will add so much latency that it will not be feasible for our applications.

bortek, 2018-11-29:
在提問時，人們甚至都不願意將文本翻譯成英文。你太無恥了。

closed 2018-11-29 by bortek

2357531789, 2018-11-30:
> 在提问时，人们甚至都不愿意将文本翻译成英文。你太无耻了。

Well, first of all, I'm sorry. This is the first time I asked a question on the wiki. I may have overlooked these details. I apologize for that!

zipray, 2018-11-30:
@bortek 
“你太無恥了”This is a very serious accusation in Chinese, maybe what you want to say is "你太懒了。"

bortek, 2018-12-01:
Ohh I am sorry guys, Google translate must have messed up something. Just wanted to say that we should use English when posting in here. Thanks.

2357531789, 2018-12-01:
> 哦，我很抱歉，谷歌翻译一定搞砸了。只想说我们在这里发帖时应该使用英语。谢谢。

Thank you very much for understanding that we should not argue on this issue. This is indeed my fault. I think we should pay more attention to the problem itself.

I want to express my point of view. Take pixhawk as an example, its function is very comprehensive and almost covers any model framework. I often say to my friends, "Play well with pix, go through the model world", but it also has its problems. Its core algorithm seems to be a little basic, and many of its functions are only able to use. It's not perfect. It's not compensated by adjusting some parameters. In my opinion, the algorithm of pix is not as good as the previous generation of A2 flight control of dji. So why are so many people obsessed with pix? That's not the unparalleled expansibility and openness of pix. That's the charm of pix.

I just want to say that delay is important for image transmission, but we can't pay too much attention to this parameter to abandon other aspects. So I hope you can think about it a little bit. If you can use the raspberry pie USB interface to collect video data, then its expansibility will be achieved. Very large improvement, and the delay of USB acquisition equipment is controllable, not too bad, which is not so intolerable for FPV voyage;

Now that I'm buying equipment, I'm going to add 4W of effort to the transmitter and use high-gain antennas on the ground to pursue long distances. As a test, the first batch of equipment I purchased was recommended by you. I expect to be able to use your updates so that my camera won't get moldy in the corner of the room.

Above I will use Google to translate and then send out, may be due to translation deviations, but also please forgive me! uuuuuuuuuu