rodizio1/EZ-WifiBroadcast #193: mavlink2 signing

Jaaaky, 2018-12-31:
Hi,
I've done some flight tests with ez-wifibroadcast. I usually use mavlink v2 protocol with signing.
The problem is that once signing is activated on pixhawk, ez-wifibroadcast is unable to get any telemetry info and I cannot establish the telemetry connection over ethernet hotspot.
I hope you can allow the use of signing as it's a real protection against someone taking control over your uav.
More on this
https://discuss.ardupilot.org/t/enabling-mavlink-2-0-and-packet-signing/25656

rodizio1, 2019-01-10:
Sorry, there are no immediate plans to support that. It could be, that it'll be supported in the future because the Mavlink parser I intend to write 'accidentally' also works with signed messages, but currently I'm not sure because I haven't looked into the details of Mavlink signing yet.

I realize, security is always a good idea, however, will this really be an issue in real-life? 

Assume you're flying with your aircraft, how many people will be near your aircraft, then getting out their notebook plus Wireshark, look at the traffic, then identify it as EZ-Wifibroadcast traffic with Mavlink inside (which is quite hard to do as EZ-Wifibroadcast is pretty much unknown and the frame format is non-standard) and then start sending Mavlink commands encapsulated in EZ-Wifibroadcast frames?

Ofcourse, if somebody knows what you're using and knows where you're flying regularly (and doesn't like you), he might do that, but it's still not trivial.

Another general problem with signing is, that (as far as I could tell by a quick google search) adds another 13 bytes to each mavlink message. For the downlink, this isn't much of an issue, for the uplink, this will increase packetloss as packetloss increases with longer wifi frames.

Jaaaky, 2019-01-14:
Actually, signing are only for uplink messages to verify the source.
It's for "authority verification", so othesr can actually connect and "see/read" the UAV info/status/location but cannot "modify or change" its settings or behavior.

Jaaaky, 2019-01-14:
If the ez-wifibroadcast just works as a general network proxy/repeater between user's PC GCS and UAV, it should work. And the ground-pi could parse the received data to get the needed info to be displayed, as it's not supposed to send any commands.

rodizio1, 2019-01-29:
>If the ez-wifibroadcast just works as a general network proxy/repeater between user's PC GCS and UAV, it should work. 

Yes, that's what one would expect, just as any internet router will route any IP packet, independently of what's the payload inside it.

Unfortunately, the Mavlink developers made a design decision that makes this virtually impossible, see here for more details: https://github.com/mavlink/mavlink/issues/834