rodizio1/EZ-WifiBroadcast #198: We use GoPro4 to connect to the raspberry pie 3B via the B101. The transmitter sets the resolution at 1080p 30fps. the video delay at the receiver is about 2 ~ 3 seconds. Why?

SnailSnake, 2019-01-18:
1. The raspberry pie sender is set to 1080P 30fps.The live video bitrate is maintained at about 13.7Mbit/s (greater than the maximum bitstream 13.2Mbit/s shown in parentheses) with a delay of about 3 seconds.
2. If the raspberry pie sender is set to 720 30fps, the live video bitrate has been maintained around 9Mbit/s (less than the maximum code stream 13.2Mbit/s shown in parentheses), the delay will return to normal, around 200ms.
3. If the raspberry pie transmitter is set to 1080p 30fps and  the raspberry pie camera is used, the live video bitrate will be at 6~7Mbit/s, with a delay of about 200ms.

Question:
1 Is the large delay caused by the fact that the real time bit stream is larger than the maximum transmission bit stream? If not, what are the reasons?
2 Using GoPro4 to connect raspberry pie via B101, at the same resolution and frame rate, why is the bitstream about twice as large as that of the raspberry pie camera?

rodizio1, 2019-01-29:
Yes, when the datastream has higher bitrate than the available max. bitrate, it is buffered and thus delayed. 

I don't know why that is and unfortunately cannot reproduce it because I don't own a B101. Reason is probably a bug in raspivid or the raspberry firmware, as the scripts inside EZ-Wifibroadcast set the raspivid bitrate parameter to the measured bitrate multiplied with the bitrate_percent config setting, i.e. much lower.

Only workaround I can think of would be reducing the bitrate_percent setting in wifibroadcast-1.txt to get a lower bitrate. Another option would be setting it to some fixed value in the /root/.profile script in the raspivid commandline.

rodizio1, 2019-02-04:
@Davidsastresas 

The "bottleneck" of 13mbit is the maximum videobitrate that can be transmitted with the configured Wifibitrate and FEC settings.

Using only raspivid on regular Raspbian probably doesn't give any delays because you're not transmitting anything via Wifi and thus nothing can get delayed.

It would be interesting to see what you get on a regular Raspbian when running the raspivid commandline with the tool pipe viever, i.e. something like:

`raspivid -w 1280 -h 720 -fps 48 -b 5000000 -t 0  | pv -i 0.5 -R > /dev/null`

That should give about 5Mbit/s (or around 625kbyte/s).

Davidsastresas, 2019-04-10:
Hello, sorry to come back this late but I left the project for a few weeks until now. 

I've come to a solution, and I have not dig enough in to the scripts for making a lot of sense of it, but I somehow have a theory. The thing is 1080p with pi camera is no problem, 1080p with picapturehd1 gets delay accumulated until it crashes.

I saw the bitrate parameter which is set to "auto" is used diferently but both by the raspivid string and some aditional argument for another script right? I suspected somehow the picapture hd1 on 1080p is outputing more bitrate that picamera at same resolution, so on the scripts on profile I eliminated the relationship between the bitrate parameter and the raspivid string. on the raspivid extra params, I set manually the bitrate, to something smaller than the available bitrate for sure. This way it works nicely, with a stunning image quality and a reasonable lag, I think I measured it at less than 200 ms glass to glass.

I don't know if that makes sense to you Rodizio, but it is finally working. Let me know if you want me to test anything in case you want to debug or something, I have the setup ready for anything you need.

Thank you very much.

Best regards.

Hanifudin95, 2023-10-14:
@Davidsastresas what parameter did you change mate? are they 

nice -n -9 raspivid -w $WIDTH -h $HEIGHT -fps $FPS -b $BITRATE -g $KEYFRAMERATE -t 0 $EXTRAPARAMS -o - | nice -n -9 /root/wifibroadcast/tx_rawsock -p 0 -b $VIDEO_BLOCKS -r $VIDEO_FECS -f $VIDEO_BLOCKLENGTH -t $VIDEO_FRAMETYPE -d $VIDEO_WIFI_BITRATE -y 0 $NICS 

what number did you put on -b $BITRATE parameter ?

Davidsastresas, 2023-10-16:
@Hanifudin95 Sorry, that was looooong ago, I didn't even remember this thread. Nowadays you are probably better using OpenHD, I think it is more active than this project.

Good luck!