rodizio1/EZ-WifiBroadcast #199: EZ-WiFiBroadcast on NVIDIA Jetson TX2

MatthewFraley, 2019-01-27:
Is there a way to utilize EZ-WiFiBroadcast to transmit from a Jetson TX2 that is using one of the wifi adapters that have been listed as working?

I am doing some image processing on a drone and want to be able to transmit the data to a ground station with high reliability.

rodizio1, 2019-01-29:
The wifibroadcast source can als be compiled on other platforms, only platform-dependent thing is the wiringpi library which can be easily removed in the makefile (If I remember correctly, only tracker.c uses it).

GiraffeRage, 2019-02-01:
Sorry for my lack of knowledge, but how would I go about compiling onto an existing Ubuntu OS on the TX2?

LalitPradhan, 2019-03-20:
Do the following:

Note: But this will disconnect you to your regular internet over wifi and eth0. Do this using the dev board with HDMI connected. Also you need to have the antennas plugged into the TX2:

sudo apt-get install hostapd dnsmasq

sudo systemctl stop NetworkManager

sudo gedit /etc/modprobe.d/bcmdhd.conf
"""
options bcmdhd op_mode=2 #add_this_in_new_line
"""

zcat /usr/share/doc/hostapd/examples/hostapd.conf.gz | sudo tee -a /etc/hostapd/hostapd.conf

sudo gedit /etc/hostapd/hostapd.conf
"""	
#add_this_in_the_top_and_comment_ssid=test
interface=wlan0
ssid=Jetson_TX2
hw_mode=g
"""

sudo gedit /etc/network/interfaces
"""
# interfaces(5) file used by ifup(8) and ifdown(8)
# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d
allow-hotplug eth0
iface eth0 inet dhcp
auto wlan0
iface wlan0 inet static
hostapd /etc/hostapd/hostapd.conf
address 192.168.8.1
network 255.255.255.0
"""

sudo gedit /etc/dnsmasq.conf
"""
#add_this_on_the_top
interface=wlan0
dhcp-range=192.168.8.20,192.168.8.254,255.255.255.0,12h
"""

sudo gedit /etc/sysctl.conf
"""
#uncomment_this
net.ipv4.ip_forward=1
"""

sudo gedit /etc/rc.local
"""
#add_this_to_the_file
iptables -t nat -A POSTROUTING -s 192.168.8.0/24 ! -d 192.168.8.0/24 -j MASQUERADE
"""

rfkill unblock 0

cat /etc/NetworkManager/NetworkManager.conf
"""
#should_look_like_this
[main]
plugins=ifupdown,keyfile,ofono
dns=dnsmasq

[ifupdown]
managed=false
"""

sudo gedit /etc/resolv.conf
"""
#add_this_after_nameserver 127.0.0.1
nameserver 8.8.8.8
"""

sudo gedit /lib/systemd/system/dnsmasq.service
"""
#add_the_last_two_lines_in_Unit_(After&Wants)
[Unit]
Description=dnsmasq - A lightweight DHCP and caching DNS server
Requires=network.target
After=network-online.target
Wants=network-online.target
"""

sudo systemctl start NetworkManager

sudo reboot

P.S. Connect to Jetson_TX2 on wifi from a separate computer and the ssh over terminal using ssh -X nvidia@192.168.8.1
To reverse all the settings just redo the steps in reverse and in /etc/resolv.conf make it nameserver 127.0.0.1

Good luck

mattgorski, 2020-06-12:
I'm looking to compile ez WiFi on the Jetson nano but we don't have an /opt/vc includes as the Pis have.  If you could provide and details on building on Jetson platforms I would be extremely grateful!!

GiraffeRage, 2020-06-12:
Jetson Nano, Nice! It has a lot of nice stuff the TX2 didn't have - like a driver for raspberry pi cameras which are CSI and allows faster imagine processing through the CPU. For your question, you will need to enable a driver for your WiFi adapter that is connected to the nano. If you don't have a wifi adapter that can go into monitor mode, you will need to get one. Compiling the adapter driver into the nano involves enabling it when you compile the OS through jetpack (if that's what it's still called). Once you have that wifi adapter working on the nano you will then have to manually install everything to get WiFi broadcast up and running. Ez WiFi broadcast is based off of an original work called WiFi broadcast. If you Google that you should find the original which includes detailed instructions on getting it working. I used a raspberry pi for the receiver since it allowed me to set things up easily and we were able to get about 2 miles with the video before it was unusable. If you need better reception than that you may want to transmit gps location data from a flight controller along with the video so you can have your receiver be pointed directly to the sender. Some things to check before start up: - Make sure you are transmitting and receiving on a wifi channel that isn't being heavily used to avoid interference.- Make sure if you are transmitting on a channel that has another adapter within range that it isn't wifi for a hospital, or fire department (the WiFi broadcast has a tendency to overpower other signals on the same channel and stop them from working)- don't transmit on 5g in the United States as it gets into cell phone band territory and can be a felony. - if you are using image recognition and tracking software, you will need to convert the video output into H264 format and then pipe than into the broadcast ( When you see a | it means pipe - something is output from one thing and immediately input into another thing)Hopefully some of that was useful, but if you have any other questions about broadcasting or drones or image processing, let me know and I can help as much as I can.

mattgorski, 2020-06-12:
@GiraffeRage Thank you for the detailed response!  I actually am already using this port of WFB https://github.com/svpcom/wifibroadcast so I have all the TX and RX side components setup (patched kernel for crda country code).  I was looking to get the wifibroadcast OSD built on the Nvidia jetson nano to broadcast osd over gst-launch to stream to a laptop and or android device.  All build instruction I can find have the CPPFLAGS set to include PI specific includes:

https://github.com/rodizio1/EZ-WifiBroadcast/blob/develop/wifibroadcast_osd/Makefile#L1

So what I am wondering and searching for is the equivalent build flags for the Jetson platform.  Thanks in advance for any experience you have building the OSD part :)