rodizio1/EZ-WifiBroadcast #204: OSD off? diversity or more antennas?

mathiasseyfert, 2019-02-19:
Hi there, 

thank you for the detailled guides, I managed to run
two Raspberry Pi3B with ALFA AWUS036NHA adapters and RpiCam.

But there are still a few questions: 

- Is there a way to donate for the project development?

- How can the OSD turned off totally to get a clean output video (without producing errors)?
  I didn't find a command in the txt-file to do that.
  (I don't need any TELEMETRY, RC oder OSD data sent)
  Would this be a way to save bandwidth?

- While having two 2.4 RC-systems in use, I need to work with WBC on 5.8 frequencies.
  Which is the best tested (diversity) WIFI adapter to get the maximum quality with 5.8
  at the range of like 300 meters?

- Is the diversity feature using a dongle with more than 1 antenna or
  does it need more than one dongle? Would it be recommended to wait for the support of ALFA AWUS1900?

Thank you in advance!

Mathias

Yes21, 2019-02-19:
Hi @mathiasseyfert ,
You can follow this [thread](https://www.rcgroups.com/forums/showthread.php?3154168-EZ-Wifibroadcast-RTL8812-RTL8814-support-thread) for AWUS1900 support.

mathiasseyfert, 2019-02-19:
Thank you, following...

rodizio1, 2019-02-20:
1. No :)

2. You can disable all the OSD elements (then technically it will still be running though). Or disable it in the /root/.profile script (see osdrx_function). No it won't save any bandwidth, as the OSD is drawn at the groundPi and not transmitted through the air

3. I'd go with Ralink RT5572 dongles, they are tried and tested. One for the Airpi and 2 for the GroundPi (or up to 5 if you want to use more antennas)

4. Both. On dongles with more than one antenna all antennas will be used and you can also use multiple dongles. AWUS1900 support will take a while and I cannot really say when I have that ready, I wouldn't wait on it.

mathiasseyfert, 2019-02-22:
Thank you @rodizio1 for the detailled comments, I will give it a try with different antennas on the new wiral lite cablecam (2.4 GHz)!
Cheers