rodizio1/EZ-WifiBroadcast #206: HELP please, pi Zero work only with 1.5  ?!

dpsoftware, 2019-03-07:
Hello, 
I downloaded the last version 1.6 RC6 and mountend on PI Zero, but the first message show 
ERROR wifobroadcast config filw not found .. and the the last message is
mount : \ boot is busy
I tried also 1.6RC5 and the problem it's the same.
With 1.5 everything work  
Someone can understand the problem ?
Thk

dpsoftware, 2019-03-07:
..Solved

rodizio1, 2019-03-21:
It checks the GPIO pins during bootup, if it uses a different configfile number, it is because the Raspberry reports those GPIO pins as "connected to ground".

Either you have one or more GPIO pins really connected to ground, or something is broken with the GPIO pins.

The error message is indeed not very clear (and contains a typo ...).

I'll see if it's possible to give the number so that it's easier to see what is going on.

dpsoftware, 2019-03-21:
Ok , but there is an information in log file to understand wich number of configuration is used  now? 
Anyway i will check the status of the GPIO pins , thank for the support

rodizio1, 2019-03-26:
No, unfortunately there is no information which file gets used as the main script in /root/.profile is not aware of the configfile copying, this is done in wbcconfig.sh.

closed 2019-03-26 by dpsoftware