rodizio1/EZ-WifiBroadcast #211: Mavlink Issues

Howie9600, 2019-06-08:
I have set up the pis as described and i am having an issue with the mavlink.  Nothing is displaying on the OSD or arriving at the GCS.  I also cannot see mavlink-router working on either pi.  is there something i am doing wrong?

staffan80, 2019-06-10:
Hi,

I am in the same boat. My take on the problem is that the RX station is not sending a request to the flightcontroller to recieve Mavlink data. I managed to get OSD data by first hooking up the 3DR dongle to a laptop with Qgroundcontrol to open up the communication then if i connected it back to the Pi rx station i was able to recieve telemetry data.

My setup is:
Arducopter FC --> 3DR dongle --> 3DR dongle --> Pi RX station

Is there any script that we can use on the Pi RX to open up the mavlink communication?
Kind regars Staffan

Howie9600, 2019-06-10:
so i have figured out more.  i have the OSD working and the TX is sending data but i cannot send data back.  only seems to be one way.  i am still investigating.

staffan80, 2019-06-16:
Hi,

Just for update. I got OSD working on the RX. The problem was that my flightcontroller did not send out the right mavlink messages as default to my 3DR dongle. The parameters that made it work in arducopter was these below. All of them where set to 0.

SR1_EXT_STAT, 2 ( 2hz for waypoints, GPS raw, fence data, current waypoint, etc)
SR1_EXTRA1, 5 ( 5hz for attitude and simulation state)
SR1_EXTRA2, 2 ( 2hz for VFR_Hud data )
SR1_EXTRA3, 3 ( 3hz for AHRS, Hardware Status, Wind )
SR1_POSITION, 2 ( 2hz for location data )
SR1_RAW_SENS, 2 ( 2hz for raw imu sensor data )
SR1_RC_CHAN, 5 ( 5hz for radio input or radio output data )

closed 2019-07-11 by Howie9600