rodizio1/EZ-WifiBroadcast #213: External GS won´t get parameters

msergiu80, 2019-07-07:
Hi. I am trying to connect Mission Planner to the groundstation, it connects on UDP and TCP but I cannot get the parameters and commands from MP do not get sent. The getting params windows show a a 5 mins wait until it gets first STAT_RUNTIME param then another long wait until maybe another param, I recently left it loading for around 30 mins and it got 4 parameters then got back to showing STAT_RUNTIME. Beside this issue I need to say it is all working. RC is working perfectly, RC=mavlink, CTS is YES but I tried N and auto. I am on 1.6 RC6 and I have AWUS036NH 2W on GS and TX, RT3070 chipset. I found a similar issue here but I do not think it is the same one since my RC=mavlink on both ends. Thank you for your help.
![2019-07-07]
(https://user-images.githubusercontent.com/7372914/60767785-7ccedd00-a0b4-11e9-8e72-3c711c2d783a.png)

msergiu80, 2019-07-07:
[debug.txt](https://github.com/rodizio1/EZ-WifiBroadcast/files/3365594/debug.txt)

[telemetry.zip](https://github.com/rodizio1/EZ-WifiBroadcast/files/3365595/telemetry.zip)

Howie9600, 2019-07-11:
The issue is with the drivers.  The card you are using needs a different driver.  The issue is the card is recieving and not transmitting.

I had this issue but after swapping to a different card the issue went away.  After a driver update the issue went away.

msergiu80, 2019-07-12:
Thank you for the answer. Playing with the system I was able to get a parameters download yesterday but I could not replicate it again afterwards. It worked one time :) Could you please tell me how you updated the driver on the EZWFB image? Or any info that could help? I cannot change the wifi card on both ends so if I can fix this issue using the same card would be great.

Howie9600, 2019-07-12:
I got it from here.  They have an image with all the drivers included.

https://www.rcgroups.com/forums/showthread.php?3154168-EZ-Wifibroadcast-RTL8812-RTL8814-support-thread

Howie9600, 2019-07-12:
Also power is a major issue with those cards.  Make sure you are powering them externally.  They need full 5v direct access or the cards reset.

msergiu80, 2019-07-12:
Thanks for the link, cards are powered separately, yes, all good there, actually the only issue I have is the bidirectional datalink, got no idea how the RC is working though. Also, the link above is for RTL8812 RTL8814, are these the same drivers needed for the AWUS036NH, Ralink 3070 chipset?

Howie9600, 2019-07-12:
It worked for me with the same card.  I had the same problem.

msergiu80, 2019-07-12:
will give it a go now and will update the results here, thanks :)

msergiu80, 2019-07-15:
Tried OpenHD and it is not working, same issue. So I am pretty much stucked. Will probably have to live without tethered Mission Planner. @Howie9600 , could you please confirm that you have RT3070 chipset on both ends and parameters load fine in external GS? Don´t know if I got it right ... :)

Howie9600, 2019-07-15:
I will check as i do use different ones now.

I know it's stupid to ask but have you set all the values in the SR1 parameter section?

msergiu80, 2019-07-15:
Yes, all values are as / wiki and beside this small issue, it is all working OK.  It would be strange to be a driver issue since RC works flawlessly. So if I can send RC messages on the link why cannot I get parameters and send commands from external GS?

msergiu80, 2019-07-15:
Ok, got it to work :)) Only Lelik´s image works. Tried everything else out there ...

closed 2019-07-15 by msergiu80