rodizio1/EZ-WifiBroadcast #221: EZ Wifi Broadcast with Navio2? Would love to hear if anyone has accomplished this.

Mysnomer0, 2019-08-24:
Has anyone been able to manually install EZ Wifi Broadcast onto Emlid's Navio2 image? I've been struggling researching this. The Navio2 image now no longer comes with the original WBC, so I was wondering if anyone has been able to combine EZ WBC with the Navio 2 image. This would be an amazing digital HD transmission and autonomous flight system in one. Any help or pointers in a good direction is greatly appreciated.

Thanks,

Mysnomer

LimitlessGreen, 2019-08-27:
Hi 
Why you're asking here (EZWB issues) for integration into the Navio 2 image?  :wink:
I thing the [Navio image builder](https://github.com/emlid/pi-gen-navio) should be a better place for this question. :octocat:

But yes, my first thought was also to integrate ArduPilot into the EZWB image and take Emlids RT Kernel instead of the stock Kernel for the RPi.

But the build process of EZWB isn't very transparent and the build script of [https://github.com/RespawnDespair/wifibroadcast-image-builder](https://github.com/RespawnDespair/wifibroadcast-image-builder) never use this repo. 

It should be possible to make this (I'm currently working on it). But both, emlid and EZWB modified and patched the kernel and it's modules.

LimitlessGreen, 2019-08-28:
Nice to hear you're working on the same thing too! :+1: 
Contribution is always welcome :tada: 
To this point, I've done only some research and played a bit around. No stuff worth to push jet.

Hopefully, good people are here. Sadly the repo not seems to be actively updated at the moment. 

I'm also not completely sure at the moment, where the best way is to start.
- [ ] Emlid's repo?
- [ ] EZWB's repo (RespawnDespair)?
- [ ] Complete new repo on the top of Raspbian?

Hopefully not the last thing.

I also noticed a few things. 
## Emlid
[This post of Emlid's forum](https://community.emlid.com/t/raspberry-pi-4-support-when/14921) suggests Emlid has plans to upgrade the image cause of RPi 4 support. But have a look at its repos. It doesn't look like activity in this direction :disappointed:
I'm not sure. Is it a good choice to wait for the new release?
First, Emlid hopefully answers to my last reply. 
Another dirty thing of emlids image builder: the repo isn't a fork of the original Raspbian. See here:
https://github.com/emlid/pi-gen-navio/issues/2
I made an own [repo](https://github.com/LimitlessGreen/pi-gen-navio) and cleaned out this issue with some advanced git-hacky stuff :shipit: 
At the moment this should be the repo to start.

## EZWB (wifibroadcast-image-builder)
Unfortunately, there was no activity in the last months. The scripts also need to be improved. For example, there is no need to delete and clone a repo again and again and again. That's a bit against the heart of git :octocat: :broken_heart:  (look at the scripts, you'll see it)

---
Here is a place for future lists
## List of parts neccesary for a navio+ezwb image
- [ ] [Emlid's RT Kernel ](https://github.com/emlid/linux-rt-rpi)(optional)
- [ ] [Emlid's RCIO-DKMS kernel module](https://github.com/emlid/rcio-dkms)
...to be continued
---

At the moment I think the best way should be to take Emlid's image builder

---

pilotnbr1, 2019-08-29:
Respawns repo got migrated into openhd. Very active development here- https://github.com/HD-Fpv/Open.HD_Image_Builder

htcohio, 2019-08-29:
Hey,

Here is the Telegram Group Link where most of the discussions take place.
If you would be interested in becoming involved, I highly recommend using
telegram it is really nice...

"""**"OpenHD 2019_RCGroups Telegram Group
This Open.HD group is intended to replace or supplement the Open.HD
RCGroups Forum which is outdated and difficult to share media.***"""
-
https://t.me/OpenHD_May_2019
-

Alse,  RCGROUPS link but, rcgroups is kind of outdated and it's difficult
to share pictures and media quickly.... But, there is still a lot of
discussion going on there.
-
https://www.rcgroups.com/forums/showthread.php?3154184-Open-HD-(BETA-TESTING)FPV-Discussion/page95
-

Justin Healey
President
HTC Communications LLC
O 440-523-1400
C 440-523-9578
www.htcohio.com

LimitlessGreen, 2019-08-29:
Nice!
I heard about it but wasn't able to find the link in the wiki :+1:

LimitlessGreen, 2019-08-29:
@htcohio: This group looks like a general user group. Is there any other group especially for developers? Maybe I would like to join.

Mysnomer0, 2019-08-30:
@pilotnbr1 @htcohio Thanks for the info! I will be following OpenHD's development now, and perhaps contributing.

archont, 2019-09-09:
I played around with this a while ago. Still on my TODO list. 

I used spvcom's wifibroadcast as base. https://github.com/svpcom/wifibroadcast - Perhaps the most technically advanced implementation, featuring multiple groundstations and a centralized controller that can rebuild frames. Also encryption and no stupid code of conduct.

Integration should be as simple as applying the 3 patches for wifi drivers. The rest is software.

Mysnomer0, 2020-01-08:
For anyone here from google searches, I've managed to pull this off. I combined svpcoms Wifibroadcast on the Navio 2 14.4.95 image. Check out svpcoms repo for more info.