rodizio1/EZ-WifiBroadcast #225: Raspberry Pi 4

3888winner, 2019-10-11:
Anyone have any luck compiling for the pi 4?

loveshuxin, 2020-01-09:
I also want to know whether the WBC supports PI 4B now ?

steveatinfincia, 2020-06-29:
It's supported in [OpenHD](https://github.com/OpenHD/Open.HD/wiki) for a couple months now, works well.

loveshuxin, 2020-06-29:
> 它已经在[OpenHD中](https://github.com/OpenHD/Open.HD/wiki)受支持了几个月，效果很好。

Thanks for your reply !

3888winner, 2020-06-30:
> It's supported in [OpenHD](https://github.com/OpenHD/Open.HD/wiki) for a couple months now, works well.

Thanks for the info, I've been trying out both OpenHD and [DroneBridge](https://github.com/DroneBridge/DroneBridge) (where pi4 is currently in testing).

RespawnDespair, 2020-06-30:
DroneBridge is a good system which targets a specific set of hardware (Pi/Camera) but does so very well, i do believe most of the architectural benefits of DroneBridge are now surpassed in Open.HD thanks to @steveatinfincia . Full disclaimer: i am heavily involved in Open.HD's development and origins, so i admit a bias to Open.HD.