rodizio1/EZ-WifiBroadcast #232: TP-Link v3 not recognized

bpjdons, 2020-04-15:
Installed EZ-WifiBroadcast on Raspberri Pi 3 and connected TP-Link v3 to it. The Wifi dongle is not recognized. Please advice what action to be taken.

steveatinfincia, 2020-06-29:
Only thing you can do is look for another set of cards I think, it sucks that they change the chips in these things I know :\

lukolszewski, 2022-04-26:
I have a couple of those cards too and I made them (barely) work by compiling and using a driver described here: https://www.hackster.io/thatiotguy/enable-monitor-mode-in-tp-link-tl-wn722n-v2-v3-128fc6

However, first the power is far less than when using v1 of the card with an atheros chip. Also the card seems stuck on slow bandwidth setting.