rodizio1/EZ-WifiBroadcast #242: 1.6RC6 release image doesn't boot on RPI Zero 2 W

lukolszewski, 2022-04-24:
RPI Zero 2 W (same BCM chip as RPI 3b) doesn't boot at all with the SD card image. It is as if there was no SD card present. Same SD card works fine on RPi3b. The card was reimaged just in case, but there was no difference. Any tips if I can make it work somehow?

lukolszewski, 2022-04-26:
I've determined it isn't working because the v4 kernels don't seem to have the device tree and dt overlays for RPi Zero2W. I've tried copying all relevant files from a newer kernel, then recompiling. I did get kernel v4 to start booting, but for some reasons my copied device tree's didn't work and it couldn't mount root fs.

So I believe the most straightforward way to get ez-wifibroadcast working on RPi Zero2W is to jbuild an image with v5 kernel (tweaking patches and config where necessary). I may try this later.

zdanek, 2022-12-09:
@lukolszewski any success with this?